MCUdude/SigmaDSP #1: Licensing

ZBindell, 2019-04-02:
Hey there, me and my buddy are working with the ADAU1701 and arduino to do a senior project and stumbled across the SigmaDSP repo. We were wondering what your plan for licensing this code was since there is a small chance we could take or project to market. Thanks!

MCUdude, 2019-04-05:
Great to hear you find this library useful! Personally, I'd like to have it under the MIT license. However much of the code is derived from the AidaDSP project. @AidaDSP is it OK for you if I license this library under MIT?

MCUdude, 2019-04-15:
@MaxPayne86 any thoughts on this?

MaxPayne86, 2019-08-23:
Hi, my library is under GPL license as stated in the official repo, so you cannot change the license to MIT as far as I know. I encourage you to release your library under the same license of mine ;).

MaxPayne86, 2019-09-04:
Hi, I'm just providing some references for my previous post. Any thoughts @MCUdude ?

> MIT licensed software can be integrated into GPL software, but not the other way around

https://en.m.wikipedia.org/wiki/MIT_License

MCUdude, 2019-09-04:
LGPL-3.0 is fine for me. According to [tldrlegal](https://tldrlegal.com/license/gnu-lesser-general-public-license-v3-(lgpl-3)), you can use this library in a closed-source project, but if you do modifications to the library it (the library) has to be kept open source.