MCUdude/SigmaDSP #5: example for dummies

yentzee, 2019-07-11:
Hi,
first of all thanks a lot for sharing your work with everyone. 
Is there any example online which documents the whole process of including buttons or encoders on an arduino to control parameters? I know that this is a dumb question resulting of a lack of knowledge. Unfortunately I am more of a musician than a person who is able to program and would just be really happy to have a useable and especially computer independent unit (i am in the process of optimizing my studio and due to the lack of space i have to think of alternatives than huge porous absorbers to get room resonances under control - i want to use the 1701 with a mic attached to an active subwoofer for a kind of low frequency noise canceling system. Unfortunately this system needs a lot of trial an error to find the right spot for it to work, prevent feedback etc and this just isn't possible with a computer attached to it)

MCUdude, 2019-07-11:
Well, the library is in fact intended for advanced tinkers who already know the Arduino framework well. I've provided some interface functions to control the ADAU1701, and then it's up to the users to adjust these variables with whatever they want.

Examples like the ones you request isn't necessary very useful to the advanced user. Personally I hate when I want to test a component (and LCD for instance) and then the example code requires me to connect a bunch of potentiometers and switches to work.

I can give you a quick example here though. Let's say you've loaded example 1 (1_Volume) and want to control the volume using a potentiometer. That's quite easy.

``` c++

// Create two signed 8-bit variables (holds -127 -> 127)
int8_t currentVolumeValue = -80;
int8_t newVolumeValue = -80;

void loop()
{
  // Map the analogRead value from 0->1023 to -80->0
  newVolumeValue  = map(analogRead(A0), 0, 1023, -80, 0);
 
  // If we have a change 
  if(newVolumeValue != currentVolumeValue)
  {
    // Write new volume value to DSP
    dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, newVolumeValue);
   
    // Prevent this if-query to be executed when there's no change
    currentVolumeValue = newVolumeValue;
  }
}
```

I haven't tested the code above on read hardware, but there's no reason why it shouldn't work.

yentzee, 2019-07-11:
Awesome, thank you so much. I suppose that is all I need to work my way through it. Again thanks a lot. :)

MCUdude, 2019-07-11:
Great! Let me know if you have any other questions!

yentzee, 2019-07-15:
Sorry to bother ou with another question. The Adau 1701 is used on a board manufactured by sure (now called wondom). There is an Eprom on the board. Can this be used the way you decribe it? It has to be connected to the iusb programmer with a second cord. When using the sure board with sigma studio just the normal way the setup is written onto it. 
Cheers,
Jens

MCUdude, 2019-07-15:
Well, I can't find a schematic for it, so I can't say for sure.

The ADAU1701 can load its firmware directly from an external EEPROM if their SDA and SCL lines are connected together, the EEPROM is added to the SigmaStudio project and the SELFBOOT pin on the ADAU1701 is pulled high. I've done it many times

yentzee, 2019-07-28:
So, I made it work finally. Now that I want to use my own setup I am getting tons of errormessages when compiling and I really don't know what to do with it. 
It took me a few weeks to get external control as well as a display with the changes I am ding working. Way too much time to stop here. What does stuff like this exactly mean? I followed your instructions 1:1
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:93:0:
sketch/SigmaDSP_parameters.h: In function 'void loadProgram(SigmaDSP&)':
sketch/SigmaDSP_parameters.h:748:122: error: no matching function for call to 'SigmaDSP::writeRegister(int, int, const uint8_t [2], int)'
     myDSP.writeRegister(CORE_REGISTER_R0_ADDR, CORE_REGISTER_R0_SIZE, DSP_core_register_R0_data, CORE_REGISTER_R0_REGSIZE);
                                                                                                                          ^
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:90:0:
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note:   candidate expects 3 arguments, 4 provided
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, const uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, const uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note:   candidate expects 3 arguments, 4 provided
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:93:0:
sketch/SigmaDSP_parameters.h:751:110: error: no matching function for call to 'SigmaDSP::writeRegister(int, int, const uint8_t [24], int)'
     myDSP.writeRegister(HARDWARE_CONF_ADDR, HARDWARE_CONF_SIZE, DSP_hardware_conf_data, HARDWARE_CONF_REGSIZE);
                                                                                                              ^
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:90:0:
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note:   candidate expects 3 arguments, 4 provided
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, const uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, const uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note:   candidate expects 3 arguments, 4 provided
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:93:0:
sketch/SigmaDSP_parameters.h:752:122: error: no matching function for call to 'SigmaDSP::writeRegister(int, int, const uint8_t [2], int)'
     myDSP.writeRegister(CORE_REGISTER_R4_ADDR, CORE_REGISTER_R4_SIZE, DSP_core_register_R4_data, CORE_REGISTER_R4_REGSIZE);
                                                                                                                          ^
In file included from /Users/jenskuehnemann/Documents/Arduino/activetrap01/activetrap01.ino:90:0:
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:65:10: note:   candidate expects 3 arguments, 4 provided
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note: candidate: void SigmaDSP::writeRegister(uint16_t, uint8_t, const uint8_t*)
     void writeRegister(uint16_t memoryAddress, uint8_t length, const uint8_t *data);
          ^
/Users/jenskuehnemann/Documents/Arduino/libraries/SigmaDSP-master/src/SigmaDSP.h:66:10: note:   candidate expects 3 arguments, 4 provided
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /Users/jenskuehnemann/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

I really don't know if I should just give up :(

MCUdude, 2019-07-29:
Can you upload your Arduino project files as a zip file, so I can ido some investigation?
Also, did you use the .sh or .ps1 script to generate SigmaDSP_parameters.h?

MCUdude, 2019-07-29:
Also, can you open the SigmaDSP_parameters.h file and replace the `loadProgram` function whis this exact one?

``` c++
void loadProgram(SigmaDSP &myDSP)
{
    myDSP.writeRegister(CORE_REGISTER_R0_ADDR, CORE_REGISTER_R0_SIZE, DSP_core_register_R0_data, CORE_REGISTER_R0_REGSIZE);
    myDSP.writeRegisterBlock(PROGRAM_ADDR, PROGRAM_SIZE, DSP_program_data, PROGRAM_REGSIZE);
    myDSP.writeRegisterBlock(PARAMETER_ADDR, PARAMETER_SIZE, DSP_parameter_data, PARAMETER_REGSIZE);
    myDSP.writeRegister(HARDWARE_CONF_ADDR, HARDWARE_CONF_SIZE, DSP_hardware_conf_data, HARDWARE_CONF_REGSIZE);
    myDSP.writeRegister(CORE_REGISTER_R4_ADDR, CORE_REGISTER_R4_SIZE, DSP_core_register_R4_data, CORE_REGISTER_R4_REGSIZE);
}
```

I belive it should work if the fuction is replaced. If so it's much easier to figure out what caused the issue in the first place. I'm pretty sure there's a bug in one of the script files though.

yentzee, 2019-07-29:
Hi, sure. That would be really nice. It is basically just the Volume script I modified to get it working with a display and a potentiometer. I had no experience with programming except flash action script 2 weeks ago. I will send the complete folder including the sigma dsp folder. I used the script on windows as the osx version wouldn't do anything :( - on a mac switching between parallels running win 10 and os x. hanks a lot for helping. Changing the sigmaDSP_parameters the way you mentioned didn't work unfortunately :(
Not sure how to attach a file using this webinterface.

yentzee, 2019-07-29:
[basstrap.zip](https://github.com/MCUdude/SigmaDSP/files/3440492/basstrap.zip)

yentzee, 2019-07-29:
[Archiv.zip](https://github.com/MCUdude/SigmaDSP/files/3440495/Archiv.zip)

yentzee, 2019-07-29:
The project is an "active Basstrap" with 3 possible strings of audio - one unedited, second for using a mic input signal, isolating the modefrequency and playing it phaseshifted (into an active sub) to cancel out by interference, third is a basic setting for a double bass array also a technique to cancel out bass on the rear wall of the studio to prevent standing waves/room modes and get a better soundgin situation on the working space in  front of the nearfield monitors. If I get it working at one point I might add another string with fiir filtering based on measurements made with rew software to even out the frequency response by the nearfield monitors. THe goal is to have quite cheap but flexible elements that can be used in between audio source (daw) and speakers to help making the room sound more linear which is a big problem in a lot of studios. there are commercial products like the e trap available but they are really expensive and without the reach of small hobby project studios. I also found schematics of completely analogue working solutions sucking the bass away instead of canceling it by interference but well I have to go on making music at one point otherwise i would build one of them as well.
Once the error messages are gone I hope that I can then just call every function of the .h file and so decide which parameters are relevant and get a dedicated hardware control (setting up delay times and leveling is absolutely crucial when setting up the active basstraps - they do work really well but it is just a pain having to connect a laptop, running sigma studio on parallels instead of just adjjusting on the unit with some potentiometers and a display) using your examples to "copy" the way you are using those functions.
Thanks a lot for helping.
Best regards,
Jens

MCUdude, 2019-07-29:
Well, there's a few things.

First, the function I posted earlier was incorrect. Use this one instead. I will fix this issue soon.
``` c++
void loadProgram(SigmaDSP &myDSP)
{
  myDSP.writeRegister(CORE_REGISTER_R0_ADDR, CORE_REGISTER_R0_SIZE, DSP_core_register_R0_data);
  myDSP.writeRegisterBlock(PROGRAM_ADDR, PROGRAM_SIZE, DSP_program_data, PROGRAM_REGSIZE);
  myDSP.writeRegisterBlock(PARAMETER_ADDR, PARAMETER_SIZE, DSP_parameter_data, PARAMETER_REGSIZE);
  myDSP.writeRegister(HARDWARE_CONF_ADDR, HARDWARE_CONF_SIZE, DSP_hardware_conf_data);
  myDSP.writeRegister(CORE_REGISTER_R4_ADDR, CORE_REGISTER_R4_SIZE, DSP_core_register_R4_data);
}
```

I will also recommend you to get the DSP working properly before adding features like Wifi and an LCD. It's much simpler to debug code where  "things that work" is removed/commented out.

Also, you can't expect the 1_Volume example to work out of the box with your Sigma Studio project. The code has to match the parameters in the generated SigmaDSP_parameters.h file. I know it's rather complicated, but the ADAU1701 along with the exported Sigma Studio files isn't exactly made for easy integration with a microcontroller. That's why this library uses a dedicated script for gathering all the useful data.

I would also recommend you to replace your volume sliders with the one with slew rate. This prevents popping noises when adjusting them.

I've edited your project slightly to get ti to compile on my computer. I replaced the `loadProgram` function, replaced the volume sliders with ones with slew rate. note that the parameter I use to adjust the volume is named `MOD_INPUTGAIN_ALG0_TARGET_ADDR`, where INPUTGAIN is derived from the name you gave it in the Sigma Studio project.

Try this project to see if you can get it to compile and run.
[Active_Basstrap.zip](https://github.com/MCUdude/SigmaDSP/files/3440920/Active_Basstrap.zip)

yentzee, 2019-07-29:
Hi, first of all - thank you so much for your help.
I ran the script but still get error messages.
But compared to before a lot less.

/Users/jenskuehnemann/Documents/Arduino/Projekte/Active_Basstrap/Active_Basstrap.ino: In function 'void loop()':
Active_Basstrap:75:21: error: 'MOD_SWVOL1_ALG0_TARGET_ADDR' was not declared in this scope
     dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, volume);

                     ^
exit status 1
'MOD_SWVOL1_ALG0_TARGET_ADDR' was not declared in this scope

Hi from Hamburg
Best regards,
Jens

MCUdude, 2019-07-29:
> Hi, first of all - thank you so much for your help.

No worries! It's important that the library is usable for other than me 😉 

The reason why you get that error is because `MOD_SWVOL1_ALG0_TARGET_ADDR` belongs to the 1_Volume example. You need to find the correct parameters in the SigmaDSP_parameters.h file and use these instead. In your case, for adjusting the input volume (with slew, that I added to your project file) you'll have to use `MOD_INPUTGAIN_ALG0_TARGET_ADDR` instead. To figure out what parameters to modify on other functions you can use the output window in sigma studio to see the changes in various registers. When you adjust the volume slider you can see that one register changes (the one that contains the volume) while the other one doesn't (because it contains the slew-rate). Hope this makes sense.

MCUdude, 2019-07-29:
Aah, sorry! I forgot to save the project before I Zipped and uploaded it! It does compile if you replace `MOD_SWVOL1_ALG0_TARGET_ADDR` with `MOD_INPUTGAIN_ALG0_TARGET_ADDR` though

yentzee, 2019-07-29:
Awesome, compiled.
Great, so now I can just start adding control to the parameters I suppose.
Just in case I want to change the setup, can I use the script within the folder you sent me to generate the .h file? Is this the corrected version?

MCUdude, 2019-07-29:
There is a bug in the Powershell script that prevents it from working properly. If you use the shell script in your project folder it's all fine. The shell script also formats the outputted code a little nicer than the Powershell script

yentzee, 2019-07-29:
Awesome. Do you have a patreon page? Or if you need some electronic music without licensing issues for one of your projects let me know. I really do appreciate your immediate help as well as the whole project.

MCUdude, 2019-07-29:
I don't have a Patreon. Everything I host here on Github is just hobby, and I don't make any money off it except for a [DIP-40 Arduino development board I sell](https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/) That goes along with [MightyCore](https://github.com/MCUdude/MightyCore). If you're interested in running "Arduino code" of other (and more capable) AVR microcontrollers this board is definitely worth its money. I used it together with an ATmega1284P when I made this library.

I'm also working on a new board I will start selling soon. It's a 42x42mm ATtiny13/25/45/85 development board with a socketed crystal, a USB to serial interface and all pins located on the same side of the board. I hope it will be popular!
[image]

yentzee, 2019-07-30:
I am fairly new to the whole thing. Are those processors it can handle capable of doing some audio-manipulatons? 
Haven'T got my wages yet but will order one the next days (regardless of their audio capabilities), if this is a good way to say thank you :)

closed 2019-07-30 by MCUdude

MCUdude, 2019-07-30:
Thanks, I just got your order! I added a small gift to your order as well. It's self-made ESP8266 breakout board, very similar to the ATtiny board I posted a 3D model of above. I'm sure you'll like it! Don't hesitate to contact me if you need assistance getting the board(s) up and running!
[image]

They (AVR microcontrollers) are not more powerful than Arduino UNO in terms of processing power, but have a few benefits. The microcontroller that comes with the (blue) board, ATmega32, is more or less similar to the ATmega328P that's in the Arduino UNO, but with more IO pins.

However, since the microcontroller is socketed you can insert bigger and more capable microcontrollers as well. If you get yourself an ATmega1284P-PU you get 16kB of RAM and 128kB of flash storage. That's eight times more RAM and four times as much flash compared to a UNO!

The AVR will never be as powerful as an ARM or an ESP32, but it's surprisingly capable for what is. is. It's not fast enough to manipulate raw digital audio such as an i2s data stream, but that's why you have a DSP, right?

yentzee, 2019-07-31:
Q¹Awesome, thank you so much.
Really looking fwd to getting deeper into it.
Just about to finish the last limp membrane traps for my studio.
Gonna start using the dsp's after that actively. Might buy 2 more (the 1401
is ridiciously cheap bought of aliexpress) to include fir filters for room
correction.
Still struggleing a bit with the whole thing as I am uncertain whetther to
use potis (would just make sense if there was a way to say "when mixer is
set to position 1 poti 1 does aaa when mixer is set to position 2 poti1
does bbb) or try to include a rotary encoder with display and menu.
Also I dont understand where to get the information on the different
parameters in the .h file (which parameter is which function as well as the
possible states for each parameter) but I think I will just try a bit and
see what happens.
Thanks a lot for the addon, really appreciate it.

Best regards from Hamburg,

Jens

yentzee, 2019-08-08:
Sorry for being a pain..
Is there any way to get information off the DSP?
It would be great to include an oled for displaying input and output levels.
Best regards

MCUdude, 2019-08-08:
There is, but this library doesn't support it yet. It's a lot of work to implement.

But can't you display the information that's already in the microcontroller? The microcontroller always "know" the current value of a block as long as the microcontroller is the only device that talks to the DSP.

MCUdude, 2019-08-08:
Are you thinking about something like a VU meter?

yentzee, 2019-08-08:
Yes, it would be great to be able to have input and output meter to prevent
clipping. Alternatively a led that blinks on clipping would be an idea. I
might be able to grab it from the comp/limiter I suppose.
Regarding settings for an eq I suppose it is possible to display encoder/
potivalues through some mathformula to show the frequency instead of
abstract values (sounds rather difficult to map a linear range to
logarithmic range).
I really hope I will finish this project. The rest of my room treatment is
done so I am desperate to start producing again.
Unfortunately with the whole setup (changed the sigma setup to one signal
line - hope to be able to control bypass / on off for each block) I might
still run out of connections to the microcontroller. Is a multiplexer worth
checking out?
Best regards,
Jens

MCUdude, 2019-08-08:
> Regarding settings for an eq I suppose it is possible to display encoder/
potivalues through some mathformula to show the frequency instead of
abstract values (sounds rather difficult to map a linear range to
logarithmic range)

Shouldn't be too difficult?
`dB_value = map(potmeter_value, 0, 1023, -80, 0);` // Map 0 to 1023 -> -80 to 0

> Unfortunately with the whole setup (changed the sigma setup to one signal
line - hope to be able to control bypass / on off for each block) I might
still run out of connections to the microcontroller. Is a multiplexer worth
checking out?

I don't think I fully understand. The DSP require only three pins from the microcontroller, no matter how many "blocks" you have in your DSP flow. However, if you have too many buttons and external things to control I often just use plain old shiftregisters. 74HC595 if I need outputs, and 74HC165 if I need inputs.

yentzee, 2019-08-08:
Unfortunately it doesn't map accordingly to frequency (or is it? I am
really bad at maths) which is logarithmic scaledue to perception of the ear
(i think?)

Yes I need 8 potis 6 buttons and 10 leds as i am way too unexperienced to
include a menu and an encoder. Looked a bit into different libraries but I
have no idea of programming and just try to look at examples and change
them to my needs.
I never knew how crazy it is having to habe so many things in your mind
when programming.
Similar to a medical diagnosis / treatment whereyou havd to have all that
background knowledge to be able to do it properly.
I was thinki g of a multiplexer to get more inputs and or outputs.
Still dont get what the different pins are for and why they aren't numbered
chronologically.
Anyway I am still confident to be able to build a box that can be connected
to every sub and be adjusted according to the needs of the room.
Also looking for a good and cheap electrect mic preamp.
Willdefinitely post results with room measurements when done :)

yentzee, 2019-08-14:
Hi,
unfortunately trying to make a new Sigma DSP setup working results in the same problems as before.
I can't execute the script under OSX (tried cding into the dir and executing using the command given in your description) and when using windows I am getting a lot of errors when using the .h file created.
:(

MCUdude, 2019-08-14:
can you post the output error of the .sh script from OSX?

MCUdude, 2019-08-14:
And/or post your entire project as a zip file so I can investigate too.

yentzee, 2019-08-14:
Unfortunately nothing happens when running the script under OS X. A new screen shows up for a second or so and that's it.
I am running El Captain if this might be an issue.
Trying to redo the sigmadsp.h in windows on parallels now.
[01.zip](https://github.com/MCUdude/SigmaDSP/files/3502197/01.zip)

MCUdude, 2019-08-14:
>Unfortunately nothing happens when running the script under OS X. A new screen shows up for a second or so and that's it. 

Have you tried making the script executable by running the following line first.
```
chmod +x DSP_parameter_generator.sh
```

 I'm on macOS Mojave (10.14) at the moment, but the script was made when I had Sierra (10.12) installed.

I tried to execute the .sh script in the zip file you provided, and it immodestly generated this file:

SigmaDSP_parameters.h
``` c
#ifndef SIGMADSP_PARAMETERS_H
#define SIGMADSP_PARAMETERS_H

#include <SigmaDSP.h>

/****************************************************************************
| File name: SigmaDSP_parameters.h                                          |
| Generation tool: AWK + bash                                               |
| Date and time: 14.08.2019 18:04:11                                        |
|                                                                           |
| ADAU1701 parameter and program file header                                |
| SigmaDSP library and its content is developed and maintained by MCUdude.  |
| https://github.com/MCUdude/SigmaDSP                                       |
|                                                                           |
| Huge thanks to the Aida DSP team who have reverse engineered a lot of the |
| Sigma DSP algorithms and made them open source and available to everyone. |
| This library would never have existed if it wasn't for the Aida DSP team  |
| and their incredible work.                                                |
|                                                                           |
| This file have been generated with the Sigmastudio_parameter_generator.sh |
| script. This file contains all the DSP function block parameters and      |
| addresses. It also contains the program that will be loaded to the        |
| external EEPROM.                                                          |
|                                                                           |
| The *_COUNT macro holds the number of addresses in memory each complete   |
| module takes.                                                             |
|                                                                           |
| The *_ADDR macro holds the current address for the module. Use this macro |
| when changing the behaviour of the modules (EQs, volume etc.).            |
|                                                                           |
| The *_FIXFT macros holds the default value of the module. Use this when   |
| restoring the default parameters.                                         |
|                                                                           |
| The DSP_eeprom_firmware[] array contains the DSP firmware, and can be     |
| loaded using the writeFirmware method in the DSPEEPROM class.             |
| When stored in the external i2c EEPROM, the firmware is automatically     |
| loaded into the DSP on boot if the SELFBOOT pin is tied to Vcc.           |
|                                                                           |
| If you want to load the DSP firmware directly without using an external   |
| EEPROM, you can simply run loadProgram() (located at the bottom of this   |
| file) where you pass the SigmaDSP object as the only parameter.           |
****************************************************************************/

/* 7-bit i2c addresses */
#define DSP_I2C_ADDRESS (0x68 >> 1) & 0xFE
#define EEPROM_I2C_ADDRESS (0xA0 >> 1) & 0xFE

/* Module Mute - Mute*/
#define MOD_MUTE_COUNT                                 1

#define MOD_MUTE_MUTENOSLEWALG1MUTE_ADDR               0
#define MOD_MUTE_MUTENOSLEWALG1MUTE_FIXPT              0x00800000

/* Module GainIn - Single SW slew vol (adjustable)*/
#define MOD_GAININ_COUNT                               2

#define MOD_GAININ_ALG0_TARGET_ADDR                    1
[... 2642 lines cut ...]
};

/* Register Default - IC 1.CoreRegister */

#define CORE_REGISTER_R4_SIZE 2
#define CORE_REGISTER_R4_ADDR 0x081C
#define CORE_REGISTER_R4_REGSIZE 2

const uint8_t PROGMEM DSP_core_register_R4_data[CORE_REGISTER_R4_SIZE] = 
{
0x00, 0x1C
};

/* Run this function to load DSP firmware directly */
void loadProgram(SigmaDSP &myDSP)
{
  myDSP.writeRegister(CORE_REGISTER_R0_ADDR, CORE_REGISTER_R0_SIZE, DSP_core_register_R0_data);
  myDSP.writeRegisterBlock(PROGRAM_ADDR, PROGRAM_SIZE, DSP_program_data, PROGRAM_REGSIZE);
  myDSP.writeRegisterBlock(PARAMETER_ADDR, PARAMETER_SIZE, DSP_parameter_data, PARAMETER_REGSIZE);
  myDSP.writeRegister(HARDWARE_CONF_ADDR, HARDWARE_CONF_SIZE, DSP_hardware_conf_data);
  myDSP.writeRegister(CORE_REGISTER_R4_ADDR, CORE_REGISTER_R4_SIZE, DSP_core_register_R4_data);
}

#endif
```

yentzee, 2019-08-14:
let me check again.

yentzee, 2019-08-14:
ok cool that worked.
gonna check if the file generated is going to work now.
:)

yentzee, 2019-08-14:
working now. Awesome. 
Thanks a lot.
:)

commit 358acea7 (2019-07-30): Fix script bug
Closes #5