MCUdude/SigmaDSP #6: Please, add support for stm32duino

Onefabis, 2020-03-05:
Hi. Since Arduino nano has not enough memory, I've got Black pill, with stm32duino support. With that configuration I could add different additional libraries like Adafruit for IPS screen for the musical center project without worrying of free memory (black pill stm32 has more that arduino nano with similar size and form). But somehow compillation finished with the errors. It doesn't like reusable functions with different type of arguments. 
Good thing is that you don't need any black pull for the test. It fires the error even without connected stm32. 
Thanks, Alex.

MCUdude, 2020-03-06:
I did some changes recently to make sure other architectures such as SAMD and STM32 should work. I'm not at my computer at the moment, så could you please post the error you're getting?

MCUdude, 2020-03-10:
I just tried to compile for the black pill board (F401Cx), and I got no errors. Reopen if you have anything to add.

closed 2020-03-10 by MCUdude

Onefabis, 2020-03-10:
Yeah, checked yesterday. No errors. Seems like I did compile in old Arduino IDE version. Laters one fires no error. Thanks, you're great!