MCUdude/SigmaDSP #7: SigmaDSP with ESP32

20163771, 2020-04-06:
Hello,

I am having problems compiling and loading any of the examples to ESP32. Arduino Uno and Nano works very well. However due to lack of memory in them I swapped to ESP32. Which now fails to compile.

`Arduino: 1.8.5 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

In file included from C:\Users\***\Documents\Arduino\libraries\SigmaDSP-master\examples\1_Volume\1_Volume.ino:13:0:

C:\Users\***\Documents\Arduino\libraries\SigmaDSP-master\src/SigmaDSP.h:145:10: error: 'void SigmaDSP::safeload_writeRegister(uint16_t, int, bool)' cannot be overloaded

     void safeload_writeRegister(uint16_t memoryAddress,      int data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }

          ^

C:\Users\***\Documents\Arduino\libraries\SigmaDSP-master\src/SigmaDSP.h:143:10: error: with 'void SigmaDSP::safeload_writeRegister(uint16_t, int32_t, bool)'

     void safeload_writeRegister(uint16_t memoryAddress,  int32_t data, bool finished);

          ^

Using library SigmaDSP-master at version 1.0 in folder: C:\Users\***\Documents\Arduino\libraries\SigmaDSP-master 
Using library Wire at version 1.0.1 in folder: C:\Users\***\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire 
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.`

I hope you can help me

MCUdude, 2020-04-06:
Thanks for reporting. I will work on a clean solution for this that works for all architectures, but for now, simply comment out line 145 in SigmaDSP.h, and it should work.

Hope you find the library useful!

20163771, 2020-04-06:
Thanks for such a quick response, it did work.

I am basing my bachelors work on ADAU1701 and ESP32. Your library part of it.

commit d027cf29 (2020-04-07): Fix compilation issue for ESP8266/ESP32
Closes #7
closed 2020-04-07 by MCUdude