MCUdude/SigmaDSP #9: No "Tutorial_0"

aprechtl, 2020-04-12:
Hello, I'm not sure if this is the appropriate place to post this, but the readme says

"This guide assumes you created your project based on the Tutorial_0 project file, which is a blank project with a few pre-configured settings."

Where is Tutorial_0?

MCUdude, 2020-04-12:
Good catch! I never created a template project to use really. I will do that. Thanks for reporting!

Meanwhile, you can just use the project file from Tutorial_1. It's the same thing. All you'll have to do is just delete the volume slider and add your own stuff.

BTW I hope you'll find the library useful!

aprechtl, 2020-04-12:
Thank you, I'm just getting started (obviously) and very excited!

commit 1f588edb (2020-10-06): Add 0_Template example
Closes #9
closed 2020-10-06 by MCUdude