MCUdude/SigmaDSP #12: Trouble with Wondom / Sure Electronics DSP (ADAU1701)

DatanoiseTV, 2020-10-23:
I've stumbled upon your library and hooked up my Wondom DSP via I2C (the connector which usually goes to the USB programmer) to an ESP32. When I run a I2C scan, I can see a device at 0x50.

Afaic, the Wondom DSP has an EEPROM and can also start from bootstraping.
Question is: why do I only see 1 device on I2C and why do the examples report

```
Pinging i2c bus...
0 -> deveice is present
2 -> device is not present
DSP response: 2
```

MCUdude, 2020-10-23:
Does the Wondom board work if you use it with SigmaStudio? Do you know if it is hardwired to be in i2c or SPI mode?
Try to run the i2c_scanner sketch to see if the DSP has a different i2c address.
https://playground.arduino.cc/Main/I2cScanner/

I recently assembled some hardware where I used an external EEPROM to store the DSP application. The EEPROM appeared in the ping, but not the DSP. The solution for me was to use a hot air station and reflow the DSP, so apparently, it was a bad solder connection somewhere.

If you still can't figure it out I suggest you contact the vendor you bought it from

DatanoiseTV, 2020-10-24:
It worked until SigmaStudio 4.1 - then the drivers were not working anymore with the ICSP from Windom.
According to the ICSP labels, it is wired in I2C mode.

A I2C scan only reveals a device at 0x50, which I guess is the EEPROM.

MCUdude, 2020-10-24:
> A I2C scan only reveals a device at 0x50, which I guess is the EEPROM.

Yes, 0x50 is the EEPROM. Then your issue is most likely a hardware issue. You can check if the DSP crystal is running if you have an oscilloscope. I've had issues with the crystal before. You should also measure to check if the core voltage is 1.8 volts (measure the DVDD pin). If you still can't figure it out your should contact the company that sold you the board.

MCUdude, 2020-11-21:
I'm closing this issue because you're experiencing a hardware issue, and that's strictly not this library's fault, nor can it do anything to fix it.

closed 2020-11-21 by MCUdude