MCUdude/SigmaDSP #13: Talk with two adau1701 simultaneously

victorbaezaevia, 2020-11-20:
Hi, im building a small mixer using a couple of adau1701 connected on the i2s and i2c bus as well,
is there any possibility to be able to talk with two (or more) adau1701 connected simultaneously on i2c bus in order to control different processing stages from each one? Regards.

xstsx, 2020-11-20:
Yes ,you can. Take a look at pins ADDR0,ADDR1 they are setting ADAU1701 i2c address.

MCUdude, 2020-11-20:
Give Your two DSPs different i2c addresses in hardware, and create two objects. One for the first DSP, and one for the second.

MCUdude, 2020-11-21:
Here's the 1_Volume example where I've also added a second DSP to the example. The second DSP is setup to have 0x35 as its i2c address.

```c++
/**********************************************|
| SigmaDSP library                             |
| https://github.com/MCUdude/SigmaDSP          |
|                                              |
| 1_Volume1.ino                                |
| This example fades the volume up and down    |
| using a slew volume adjustment slider.       |
| See the SigmaStudio project file if you want |
| to learn more, tweak or do modifications.    | 
|**********************************************/

#include <Wire.h>
#include <SigmaDSP.h>
#include "SigmaDSP_parameters.h"

SigmaDSP dsp(Wire, 0x34, ADAU1701 /*,12*/);
SigmaDSP dsp1(Wire, 0x35, ADAU1701 /*,12*/);

void setup() 
{  
  Serial.begin(9600);
  Serial.println(F("SigmaDSP 1_Volume example\n"));

  Wire.begin();
  dsp.begin();
  dsp1.begin();

  delay(2000);

  
  Serial.println(F("Pinging i2c lines...\n0 -> deveice is present\n2 -> device is not present"));
  Serial.print(F("DSP response: "));
  Serial.println(dsp.ping());
  //Serial.print(F("EEPROM ping: "));
  //Serial.println(ee.ping());
  
 
  // Use this step if no EEPROM is present
  Serial.print(F("\nLoading DSP program... "));
  loadProgram(dsp);
  loadProgram(dsp1);
  Serial.println("Done!\n");

  // Comment out the three code lines above and use this step instead if EEPROM is present
  // The last parameter in writeFirmware is the FW version, and prevents the MCU from overwriting on every reboot
  //ee.writeFirmware(DSP_eeprom_firmware, sizeof(DSP_eeprom_firmware), 0);
  //dsp.reset();
  // dsp1.reset();
  //delay(2000); // Wait for the FW to load from the EEPROM
}

void loop() 
{
  // Slowly adjust the volume down to -50dB
  Serial.println(F("Fading down"));
  for(int8_t i = 0; i > -51; i--)
  {
    dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i, 12); // (First volume slider address, dB value, optional slew value)
    dsp1.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i, 12); // (First volume slider address, dB value, optional slew value)
    delay(200);
  }
  
  // Slowly adjust the volume up to 0dB
  Serial.println(F("Fading up"));
  for(int8_t i = -50; i < 1; i++)
  {
    dsp.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i); // Slew rate is default 12
    dsp1.volume_slew(MOD_SWVOL1_ALG0_TARGET_ADDR, i); // Slew rate is default 12
    delay(200);
  }
}
```

closed 2020-11-21 by MCUdude