MCUdude/SigmaDSP #14: 48 FS question

Onefabis, 2021-01-01:
Hi, I see FS = 48 in parameters.h, if that is a constant, does it mean that all the sigma studio blocks in your code are calculate coefficients for projects generated from sigma in 48kHz? If I'm wrong in that assumption, how and where this coefficients recalculated for different sampling rates, like 96kHz, 192kHz, etc? I'm asking because if I change parameters in sigma studio, for instance in Volume slew, I don't see any changes in values while change project sampling rates, but I see some changes in EQ Second Order, for instance. Is I change project sampling rate, I see different values in capture window in sigma. So, please, shed the light what happening under the hood of your code, is there anything what re-calculate those values for different sampling rates?

MCUdude, 2021-01-01:
Hi!

>  I see FS = 48 in parameters.h, if that is a constant, does it mean that all the sigma studio blocks in your code are calculate coefficients for projects generated from sigma in 48kHz?

The sample rate (FS) is defined here as a constant:
https://github.com/MCUdude/SigmaDSP/blob/1c3b88ef445316ddb379690be9e632877f1e37bf/src/parameters.h#L5-L7

At the moment it is not possible to change the sample rate without changing the `#define FS` in the parameters.h file. I'll see if the sample rate can be passed to the constructor instead to improve code portability.

The sample frequency, `FS` are used in `audioDelay()`, `EQfirstOrder()`, `EQsecondOrder()`, `toneControl()`, `stateVariable()`, `compressorRMS()` and `compressorPeak()`.

To be honest I've never needed to use higher sample frequencies than 48 kHz because my hearing is limited by the Nyquist-Shannon theorem 😉

Onefabis, 2021-01-01:
Thanks for the answer. Yes, agree with you, I also believe that higher sample rates gives no hearable difference. I simply avoid resampling in any stage of my audio project. I just have USB transport, that can change its output FS, depending on the sample rate of the audio file that played. So this transport give highs and low on some pins, telling what FS it's working on right now. And depending on this pins I change project sampling rates and load it from arduino to DSP. But I get all that data in capture window in sigma and store it in array right now. That's not elegant solution, but it's work stable with different sampling rates

MCUdude, 2021-01-01:
> Yes, agree with you, I also believe that higher sample rates gives no hearable difference. I simply avoid resampling in any stage of my audio project.

That is very understandable.

> So this transport give highs and low on some pins, telling what FS it's working on right now. And depending on this pins I change project sampling rates and load it from arduino to DSP

So you have multiple DSP firmwares compiled with different sample frequencies you load using the microcontroller? If so, you _could_ create multiple objects of the SigmaDSP library instead.

> But I get all that data in capture window in sigma and store it in array right now. That's not elegant solution, but it's work stable with different sampling rates

As long as it works, it's fine. I'm sure this library could be tweaked to do exactly what you want it to, but you'll have to know the source code and dive into it. What "DSP blocks" are you modifying in your project(s), and do you use safeload_write to write the data in the array to the DSP?

Onefabis, 2021-01-02:
> I'm sure this library could be tweaked to do exactly what you want it to, but you'll have to know the source code and dive into it. 
I can do that, just give me direction where to start.

> What "DSP blocks" are you modifying in your project(s), 
[image]
Here is my project for headphone amplifier. First Second Order EQ corrects my headphones, based on this page https://www.reddit.com/r/oratory1990/wiki/index/list_of_presets
Second one is 5 line EQ, that tune up manually by encoder, so that changes displayed on the screen and the last one is loudness, it also can be turned on/off and those values changed on every +/-10dB to make a simple correction of loudness 
[image]
So I think tha main purpose of DSP is EQ and phase coddection with EQ, delay and FIR (but 1701 is weak for FIR filter). Even volume control is temporary solution since I'd like to do it with PGA2311. 

> and do you use safeload_write to write the data in the array to the DSP?
Recently I use files from that page https://wiki.analog.com/resources/tools-software/sigmastudio/tutorials/microcontroller
It can work with any Sigma100-200-300-350 devices, so since I have 1701, 1452, 1467 I use it in every personal project.
I don't see any obvious method in SigmaStudioFW.h that work like a Safeload_write,maybe I'm wrong.

MCUdude, 2021-01-02:
Cool, but what do you need this library for then?

Onefabis, 2021-01-02:
As I sad this way I need to save data in array from capture window, for all sampling rates: 44.1, 48, 88.2, 96, 176.4, 192 It take too much time even if I use my own python script to convert that data to array. And in case if I change something in EQ, like Q-factor, I have to capture it again :( So that's why I need it in formula, also I don't shure that Safeload_write work in SigmaStudioFW.h. Perhaps it's not better way to change registers, because it may cause hearable clicks. In some cases I can hear it.

MCUdude, 2021-03-20:
The sample rate is no longer hard-coded into the library, so you're free to change it in the constructor.

closed 2021-03-20 by MCUdude