MCUdude/SigmaDSP #15: Using Mighty Core to create an ATmega32 bootloader for Adau1701

Aragorus, 2021-06-18:
Would it be possible to program an ATmega32 (using MightyCore) as a bootloader for Adau1701?
I'm trying to bring to life a pair of Powersoft DigiMod 3004PFC4 amps with Powersoft DSP-D cards.
Powersoft offers a programming board which is both expensive and not supported in SigmaStudio any more.
The architecture of the DSP-D looks like this:
[image]

Seems like all the important pins are exposed:
[image]

Thank you!

MCUdude, 2021-06-18:
Hi!
There is no such thing as a bootloader for the ADAU1701. You can either let it boot directly from a separate i2c EEPROM, or the microcontroller can load its program on boot. All the examples in this repo expose both features, so you're free to choose. I've used the ATmega32 along with ADAU1701 many times, so there's no reason why it shouldn't work.

But are you sure the DSP isn't in self boot mode and the DSP boots directly from the i2c EEPROM? This would make the most sense since the ATmega32 is a bit limited in terms of flash memory to hold the DSP program.

BTW why would you connect the CS4398 DAC to the microcontroller and not the DSP? Looks like a drawing error to me.

May I ask what delay chip you're using? I've only used the TPA5050 with the ADAU1701, but I'm sure other works great as well.

Aragorus, 2021-06-18:
I got the info from powersoft's User Guide:
https://www.powersoft.com/wp-content/uploads/2019/01/powersoft_DSP-D_Uguide.pdf

In terms of self boot - I've tried to add Eeeprom in Sigma Studio, but it doesn't find any.
I believe powersoft did it in this way so that they can recall presets with a button.

Again, it's not my design. It's a DSP card that sits in the amp itself. And since it's closed source the best way to make it work would be to reprogram the Atmega32

[image]

MCUdude, 2021-06-18:
Personally, I would desolder the ATmega32 and insert an ATmega1284 instead. Then you have tons of space to store your DSP program(s). But you can get away with the ATmega32 too.

May I ask what the original price for the board was? The price of the hardware itself isn't much at all.

What happens if you install MightyCore and upload the i2c scanner sketch? Does the DSP, delay IC, and EEPROM show up as separate devices with unique addresses?

Aragorus, 2021-06-18:
I got a pair amps with these DSP cards inside for around 13000 NOK a couple of years ago. The amplifier is capable of delivering 4x2000w in 4 ohm, or so they say.
I'll try the MightyCore Scanner and let you know!

Aragorus, 2021-06-18:
I wasn't able to upload the code, but in the process of trying decided to ground the store preset pin to ground and indeed it stored the preset. However it only worked if the sigma usbi was disconnected