MCUdude/SigmaDSP #16: Reducing memory usage

djamps, 2021-07-05:
Is there a way to omit the DSP program data from the c headers?   I'm currently self booting so it's really not needed, and the arduino's program memory is on a tight budget given the existing program in place.   Thanks in advance.

MCUdude, 2021-07-05:
Hi!

If you're not using the microcontroller to flash the EEPROM, you can shave off quite a lot.

On my computer, the *1_Volume* test sketch compiles down to 16292 bytes on an Arduino UNO. If I comment out `loadProgram(dsp);`, the sketch size drops significantly and is now only 6738 bytes.

The large arrays in the SigmaDSP_parameters.h file don't get included by the compiler if they aren't in use.

djamps, 2021-07-05:
Thanks, that helps alot!   Do you have any thoughts on using self booting + MCU?   We technically have two bus masters in this scenario.   It seems to work ok but I couldn't find any info in the datasheet on exactly how to load the DSP program outside of self boot.

MCUdude, 2021-07-05:
> Do you have any thoughts on using self-booting + MCU?

This is how I've usually done it. The DSP is only a master for a few hundred milliseconds on power-up, so as long as you give the DSP the time it needs you're all good.

> It seems to work ok but I couldn't find any info in the datasheet on exactly how to load the DSP program outside of self boot.

It's quite easy to load the program using a microcontroller on boot; at least with this library. All the test sketches in this library show you how it can be done using `loadProgram(dsp)`. And all the library source code is here if you really want to go down the rabbit hole!

It takes up a lot of space indeed, but it gives you greater flexibility if you have a microcontroller with a lot of flash memory. An ESP32 could for instance have several DSP programs in flash, and load the appropriate one when needed.

djamps, 2021-07-05:
Thanks so much for the detailed info and quick replies.   I'll consider this case closed :).

closed 2021-07-05 by djamps