MCUdude/SigmaDSP #17: Writing EEPROM bricks the DSP

djamps, 2021-07-07:
Hello,

I have had success so far except for writing new EEPROMs.   While using ee.writeFirmware(), the write appears a success (it no longer attempts writes until version is increased), however the DSP now fails to self boot.   I have to hook up a EZ-USB and rewrite the ROM from Sigma Studio to restore the program.   I made 100% sure nothing else on the I2C bus is interfering and DSP is held in reset during write.

I tried to dump copies of the good and bad EEPROM contents but while Sigma Studio displays them without issue, it refuses to write them to  files for some reason.   I have attached my project files, generated from Sigma Studio 4.6.

[ghetto.zip](https://github.com/MCUdude/SigmaDSP/files/6778215/ghetto.zip)

MCUdude, 2021-07-07:
Thanks for reporting. It may be the power shell script that is causing issues. Can you try to replace your SigmaDSP_parameters.h with this file and try again?
[SigmaDSP_parameters.h.zip](https://github.com/MCUdude/SigmaDSP/files/6778593/SigmaDSP_parameters.h.zip)

MCUdude, 2021-07-07:
If this doesn't work, another thing you should try is a very basic program such as the 1_volume test sketch. This is a pre-compiled DSP project that I know works.

djamps, 2021-07-07:
1_volume works.

MCUdude, 2021-07-07:
> 1_volume works, even after recompiling it completely. trying some more tests now.

That good. This means that the files that SigmaStudio spits out works with the script.

The code that flashes the EEPROM is an entirely separate thing and has little to do with the `loadProgram(dsp)` function. If there where something wrong with the program loading, I would expect it to be related to `loadProgram(dsp)`, not the EEPROM flashing. At the moment I only have a Mac available, so I'm not able to look at your SigmaStudio project at the moment. But try to remove some of the program blocks to see if you can make the problem go away. You can also use SigmaStudio to compare the contents of a "good" EEPROM with a "bad" one to see if there are any differences. There shouldn't be any, since E2prom.Hex is just converted into a C style array, stored in SigmaDSP_parameters.h, and written to the EEPROM.

djamps, 2021-07-07:
I tried writeFirmware() for 1_volume but it seems to again brick the DSP/EEPROM.   

I've attached my modified version of the compiled sketch (I needed to add a tone generator and control lines for the DSP on my custom board).

To be clear, this works fine using loadProgram().   Unfortunately due to other technical issues, I need to use an EEPROM.

[1_Volume.zip](https://github.com/MCUdude/SigmaDSP/files/6778880/1_Volume.zip)

MCUdude, 2021-07-07:
But you're able to get it to work if you flash the EEPROM with the EZ USB programmer? Try to compare the EEPROM content that's working with the one that's not working.

djamps, 2021-07-07:
[image]

Attached are the non-working (left) and working (right).   I still cannot get sigma to save the eeprom hex, but due to the small program it fits in screen shot.   The FF's at the end of the non-working continue to around half the 128kbit EEPROM memory.

djamps, 2021-07-07:
It looks like I've got two copies of E2Prom.Hex (original and modified).   I am deleting my last post and re-testing.

djamps, 2021-07-07:
I was successful in burning the EEPROM with 1_Volume, the only differences in the final dump were FF's instead of 00's filling the rest of the EEPROM.   But it works.

However, it still doesn't work for my larger sketch.   Once I figure how to save the EEPROM to file, I can compare.   As of now, sigma studio refuses to save files using the "Upload to file" function.   It will only display them on screen with no cut/paste ability.   If you know another way to dump the EEPROM let me know?

MCUdude, 2021-07-07:
> If you know another way to dump the EEPROM let me know?

You can dump the EEPROM using an Arduino sketch and print it to the serial monitor, correctly formatted with for instance printf.

djamps, 2021-07-07:
I may have found the issue, and it's in the board design.   I stupidly tried to use A6 as a digital out on the 328P (for DSP writeback, which I don't even need), and as a result it was controlling D6 instead, which was pulling WP low and stopping the boot process.    Once I confirm these initial findings, I'll be close this issue in shame :).

djamps, 2021-07-07:
Confirmed design issue.  Everything is working as expected now that the WP pin state is known/good.  Excellent work on this library, it saves me alot of hassle with embedded DSP programming.

closed 2021-07-07 by djamps

MCUdude, 2021-07-07:
Great to hear you figured it out! I've used this library with some massive DSP programs, and never had any issues regarding flashing or controlling the DSP blocks. Great to hear you find the library useful! I've spent quite a lot of time creating it, so it's nice to hear it is being used by others.

The library is actually quite extendable, so if you crack an algorithm (shame on AD for making it so difficult!), it's pretty straightforward to implement the algorithm as a library function.

djamps, 2021-07-07:
Indeed, it's written very concisely and easy to follow, I already extended it for controlling DC sources.  I'm able to store an entire DSP program (1000+ instructions) and hundreds of lines of additional code on a 328P, so the efficiency shows!   I'll send you a pull request soon!   Thanks again for the great work.