MCUdude/SigmaDSP #18: demux

ara9329, 2021-07-08:
Hey 
great work I love to work with it 
at the moment with an ttgo t-display and a adau1701 as preamp xover eq 3d
volume and input select  MUX works perfect.

i am struggling with the demux syntax
its 1 2xn so 2 in 3 out

     dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW10_ADDR, 1, 4); 
        dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW11_ADDR, 0, 4);
        dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW12_ADDR, 0, 4);
Can you explain howto use the parameters?

Thanks

MCUdude, 2021-07-08:
Can you post a screenshot so I can see what the mux looks like in the SigmaStudio GUI. IIRC there are two types of demuxes that need to be controlled in two separate ways. One with index, another one without.

ara9329, 2021-07-08:
[image]

ara9329, 2021-07-08:
Thanks for reacting so fast

closed 2021-07-08 by ara9329

MCUdude, 2021-07-08:
I don't have a Windows computer at hand, so I can't help you directly by looking at the SigmaStudio project file.

However, if you look at the SigmaStudio capture window you'll see that there are probably several registers changing when you change the demux state. This is the SigmaDSP::demux function:

https://github.com/MCUdude/SigmaDSP/blob/3b2f61dc13eeb872a248c108a01451be624b24c0/src/SigmaDSP.cpp#L102-L117

I'll assume you'll refer to the one with three possible combinations, all the way to the right. This one has three indexes, not four. If you look in the capture window after you've compiled, you can see that it sends a sequence of three bytes depending on which position is set.

I'll guess the syntax would be something like `dsp.demux(start_address, 1, 3); ` to select the middle one of the three.

ara9329, 2021-07-08:
thanks again:

Ill give another try
although I think ive tried that already.

ara9329, 2021-07-08:
void chng_output(int t){  //change output channel
 switch (t){
    //case  0: tt=1; break;
    case  1:                                                  //AN1
        dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW10_ADDR, 1, 3); 
        //dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW11_ADDR, 2, 3;
        Serial.println("out=2");
        break;
    case  2:                                                  //DAC1
        dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW10_ADDR, 2, 3); 
        Serial.println("out=3");
        break;    
    default:                                                   //AN1 
        dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW10_ADDR, 0, 3); 
        //dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW11_ADDR, 0, 3);
        //dsp.demux(MOD_2XN_1_ALG0_STEREODEMUXSLEW12_ADDR, 0, 3);
        Serial.println("out=1");
        break;
}
This what i have tried
i hear output when downloaded
when the output is changed  i can hear the slew
but when selecting the first again I do not get the output 
back

ara9329, 2021-07-09:
I have made a copy of your volume example
and changed it for testing de Demux
[volume_demux.zip](https://github.com/MCUdude/SigmaDSP/files/6789305/volume_demux.zip)