MCUdude/SigmaDSP #22: Crossover

rudiars, 2021-11-25:
hi, I need help how to control ADAU1701 crossover ?
thanks for respond..

MCUdude, 2021-11-25:
There are several crossovers. Which are you thinking about?

rudiars, 2021-11-28:
I need 2nd order 3 way crossover sir, thanks for respond

rudiars, 2022-02-15:
please help...

MCUdude, 2022-02-15:
Why not just use three 2nd order filters, and control each of them individually?

djamps, 2022-03-19:
@MCUdude hey dude!   Any chance to control Nth order 1 channel filters?   2nd order isn't quite enough for my application.   Thanks again for your awesome work!

MCUdude, 2022-03-19:
@djamps Everything can be controlled, but you'll have to figure out the algorithm yourself, since this library currently doesn't support it. If you don't want to go down that rabbit hole, you can always just connect two 2nd order filters in series.

djamps, 2022-03-19:
If it helps anyone, here's a trick to have adjustable crossovers without having to reverse engineer the algo:
https://ez.analog.com/dsp/sigmadsp/f/q-a/64925/sure-dsp---crossover-function

closed 2022-12-17 by MCUdude