MCUdude/SigmaDSP #26: Question to safeload_write

fmvoxasoft, 2022-04-14:
Hello,

I want to write some own function for controlling the blocks into the dsp,bBut can't test this at the moment, so I am asking here before.
I don't understand correctly how to use the data block in safeload_write.

As example the mute-block. SigmaStudio says:
[image]
which of them is now the correct method?
```
  dsp.safeload_write(startMemoryAddress, 0x80 );
  dsp.safeload_write(startMemoryAddress, 0x00, 0x80, 0x00, 0x00 );
  dsp.safeload_write(startMemoryAddress, 0x00800000 );
```
Thanks and greet work with this library!
fmvoxa

fmvoxasoft, 2022-04-14:
I see in the function for the `mux`, that there is the index (0-N) as a parameter, this isn't in hexadecimal. So it is also possible to use 0 or 1 for the mute example?

MCUdude, 2022-04-14:
The screenshot from SigmaStudio isn't correct, because you haven't compiled the project. That's why the block is shown with address 0x0000.

Each address takes a 32-bit number. For a mute block with only one address, this is the correct way:
```
dsp.safeload_write(startMemoryAddress, 0x00800000); //Enable
dsp.safeload_write(startMemoryAddress, 0x00000000); //Disable
```

fmvoxasoft, 2022-04-14:
Thanks for the quick replay!

> The screenshot from SigmaStudio isn't correct, because you haven't compiled the project.

It is compiled. In the `SigmaStudio_parameter.h` is the address 0 too. It is the first block behind the input-block.

MCUdude, 2022-04-14:
Sorry, I assumed it wasn't compiled.

I'll assume you got it working? If you need more examples of how you can utilize `safeload_write`, have a look at [SigmaDSP.cpp](https://github.com/MCUdude/SigmaDSP/blob/master/src/SigmaDSP.cpp)

fmvoxasoft, 2022-04-16:
Yeah, it is working!

Here is the function, for everyone who need it too:
```
void setMute(bool status)
{
  Serial.print("Mute: ");
  Serial.println(status ? "True" : "False");
  uint16_t startMemoryAddress = MOD_MASTERMUTE_ALG0_MUTEONOFF_ADDR;
  uint32_t mute = status ? 0x00000000 : 0x00800000;
  dsp.safeload_write(startMemoryAddress, mute);
}
```

MCUdude, 2022-04-16:
Great, thanks for sharing!

closed 2022-04-16 by MCUdude