MCUdude/SigmaDSP #27: I2C doesn't communicate with adau1401

6Resh6, 2022-04-30:
I've tried to upload my sigmaDSP project to adau1401 AliExpress module in both selfboot and direct (from MCU) boot modes. The library returning that the device doesn't present (code 2).
I've made tests on teensy 4.0, the standard wire library didn't work (was returning code 4), then I've tried special library from teensy official webpage, and it's returning something like connection time over with the same code 4.
The I've switched to esp8266 using standard wire library and it's still nothing just code 2, device is not present...

Is there any ways to find out what's wrong?

Thank you

MCUdude, 2022-04-30:
Do you have a link to the board you bought from Aliexpress?

Can you try to run the [i2c_scanner](https://playground.arduino.cc/Main/I2cScanner/) sketch and post the output here? Also, are there external pullup resistors on the i2c lines on your board?

6Resh6, 2022-04-30:
Here is the board

https://a.aliexpress.com/_vY3uRv

Yes there are pullups 6k on the board
I'll run the scanner, thank you.
Also many I've found the answer to the problem on one forum. I'll update if it'll solve it.
Thank you for quick response!

6Resh6, 2022-04-30:
This is scanner results:
I2C device found at address 0x34  !
I2C device found at address 0x50  !

This what I've got from my test sketch:
[image]

MCUdude, 2022-04-30:
Ok, so the board appears when scanning the bus... 0x34 is the DSP, and 0x50 is the EEPROM. Can you post your code, including the contents of the SigmaDSP_parameters.h file?

6Resh6, 2022-05-01:
Main file:

/**********************************************|
  | SigmaDSP library                             |
  | https://github.com/MCUdude/SigmaDSP          |
  |                                              |
  | 0_Template.ino                               |
  | This example is ment to be used as a         |
  | template for your own projects. It provide   |
  | no functionality other than connecting the   |
  | two analog inputs directly to output 0 and 1 |
  |**********************************************/
#define DEVICE_ARCHITECTURE_IC_1                  "ADAU1401"
#define WP D5

// Include Wire and SigmaDSP library
#include <Wire.h>
//#include <i2c_driver.h>
//#include <i2c_driver_wire.h>
#include <SigmaDSP.h>

// Include generated parameter file
#include "SigmaDSP_parameters.h"

// The first parameter is the Wire object we'll be using when communicating wth the DSP
// The second parameter is the DSP i2c address, which is defined in the parameter file
// The third parameter is the sample rate
// An optional fourth parameter is the pin to physically reset the DSP
SigmaDSP dsp(Wire, DSP_I2C_ADDRESS, 44100.00f /*,12*/);

// Only needed if an external i2c EEPROM is present + the DSP is in selfboot mode
// The first parameter is the Wire object we'll be using when communicating wth the EEPROM
// The second parameter is the EEPROM i2c address, which is defined in the parameter file
// The third parameter is the EEPROM size in kilobits (kb)
// An optional fourth parameter is the pin to toggle while writing content to EEPROM
DSPEEPROM ee(Wire, EEPROM_I2C_ADDRESS, 256, LED_BUILTIN);

void setup()
{
  Serial.begin(115200);
  //  Serial1.begin(115200);
  Serial.println(F("SigmaDSP 0_Template example\n"));

  pinMode(WP, OUTPUT);
  digitalWrite(WP, HIGH);
  delay(1000);
  digitalWrite(WP, LOW);

  dsp.begin();
  ee.begin();

  delay(2000);
 Serial.println(DSP_I2C_ADDRESS,HEX);
  Serial.println(EEPROM_I2C_ADDRESS,HEX);
  Serial.println(F("Pinging i2c lines...\n0 -> deveice is present\n2 -> device is not present"));
  Serial.print(F("DSP response: "));
  Serial.println(dsp.ping());
  Serial.print(F("EEPROM ping: "));
  Serial.println(ee.ping());

  // Use this step if no EEPROM is present
  Serial.print(F("\nLoading DSP program... "));
  //   loadProgram(dsp);

  // Comment out the three code lines above and use this step instead if EEPROM is present
  // The last parameter in writeFirmware is the FW version, and prevents the MCU from overwriting on every reboot
  ee.writeFirmware(DSP_eeprom_firmware, sizeof(DSP_eeprom_firmware), 1);
  digitalWrite(WP, HIGH);
  delay(1000);
  dsp.reset();
[... 172 lines cut ...]
const uint8_t PROGMEM DSP_hardware_conf_data[HARDWARE_CONF_SIZE] = 
{0x00, 0x18, 0x08, 0x00, 0x00, 0x02, 0x44, 0x02, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01};

/* Register Default - IC 1.CoreRegister */
#define CORE_REGISTER_R4_SIZE 2
#define CORE_REGISTER_R4_ADDR 0x081C
#define CORE_REGISTER_R4_REGSIZE 2
const uint8_t PROGMEM DSP_core_register_R4_data[CORE_REGISTER_R4_SIZE] = 
{
0x00, 0x1C};

/**
 * @brief Function to load DSP firmware from the microcontroller flash memory
 * 
 * @param myDSP SigmaDSP object
 */
inline void loadProgram(SigmaDSP &myDSP)
{
  myDSP.writeRegister(CORE_REGISTER_R0_ADDR, CORE_REGISTER_R0_SIZE, DSP_core_register_R0_data);
  myDSP.writeRegisterBlock(PROGRAM_ADDR, PROGRAM_SIZE, DSP_program_data, PROGRAM_REGSIZE);
  myDSP.writeRegisterBlock(PARAMETER_ADDR, PARAMETER_SIZE, DSP_parameter_data, PARAMETER_REGSIZE);
  myDSP.writeRegister(HARDWARE_CONF_ADDR, HARDWARE_CONF_SIZE, DSP_hardware_conf_data);
  myDSP.writeRegister(CORE_REGISTER_R4_ADDR, CORE_REGISTER_R4_SIZE, DSP_core_register_R4_data);
}
#endif

MCUdude, 2022-05-01:
The DSP and EEPROM show up when scanning the i2c bus, so there's something with your sketch that prevents the DSP from working properly.

Can you try to comment on the following part in your sketch?

```c
  pinMode(WP, OUTPUT);
  digitalWrite(WP, HIGH);
  delay(1000);
  digitalWrite(WP, LOW);
```

Also, does the EEPROM show up as present (code 0) in your sketch?

Last question, What are you using a sampling rate of 44.1 kHz? It is the internal registers and the external crystal that determines this.

* If the crystal is a 12.288 MHz one, use 48 kHz. 
* If the crystal is an 11.2896 MHz one, use 44.1 kHz.

MCUdude, 2022-05-01:
For reference, here is the schematic for an ADAU1701-based design I've made that works perfectly.

The DSP self boots by loading its firmware from the EEPROM, and a microcontroller is adjusting the parameters in the DSP at runtime. (The TPA5050 is irrelevant, it's just there to add programmable audio delay).

[image]

6Resh6, 2022-05-01:
Ok I'll change frequency and try again.
Both the memory ping and DSP ping response with code 2, not present.
The WP output added to control memory write protect pin, according to one forum about my specific DSP module if not to pull it to the ground you cannot write down the program to the memory....

6Resh6, 2022-05-02:
This is schematics of a module similar to my AliExpress board (with some small differences)
#
[image]
Looks almost the same like your if I'm right...
Only the problem that I have no options to change selfboot move to direct...

MCUdude, 2022-05-03:
I'm very confident I found the issue, and it's 100% my fault.

The 0_Template example lacks a crucial line of code; `Wire.begin()`!
The other examples include this line, but not the template for some reason.

Try adding `Wire.begin();` _before_ calling dsp.begin and ee.begin and see if you can get it to work now.

6Resh6, 2022-05-03:
Great! It works! Thank you for your help!

MCUdude, 2022-05-03:
Great, have fun!

closed 2022-05-03 by MCUdude