MCUdude/SigmaDSP #31: Use 2 output ADAU1701

Thread09, 2022-07-30:
Hi,

I'm using an ADAU1701 and I can't find how to set different parameters on output 1 and output 2.

Do I need to use writeRegister or something is already develop ?

Thanks

MCUdude, 2022-07-30:
> I'm using an ADAU1701 and I can't find how to set different parameters on output 1 and output 2.

What are you trying to achieve? What parameters do you want to set? You can do all permanent settings in SigmaStudio, and it will be present when you export the project.

MCUdude, 2022-08-02:
>  I want to set different parameters (like band pass, volume, ..) for output 1 and output 2.

See the screenshot below. You don't make any changes to the output blocks themselves. You connect various blocks to them.

The screenshot below is for an amplifier that takes a mono signal as input. Output 0 and 1 are for a speaker amplifier, and output 3 and 4 are for a headphone amplifier.

You should play around with SigmaStudio and get familiar with it. This library is designed to tweak the parameters of the blocks you've pulled into your SigmaStudio program.

[image]

closed 2022-08-29 by MCUdude