MCUdude/SigmaDSP #33: noob guide

mihaigabriel620, 2022-09-12:
Hello if you have the time is it possible to make a video that explains it all

Thanks

Ps if you cant 

can you send me the code for arduino to use the adau1701 for 3 way speaker system 

and to use  the potentiometers that are on the wondom board ( bass,mid,high,gain )

mihaigabriel620, 2022-09-12:
here is the scenario i have a geneva xl speaker that the amp is broken and i have to build a new one and i want to use the adau1701 as 3 way crossover for the tweeter,midbass and bass speakers

and also with the 4 potentiometers on the bord to control things like ; bass boost, mid boost, tweeter boost and the gain.

if you can do the code for me that whould be appreciated.

Thanks

mihaigabriel620, 2022-09-12:
[image]

here is the speaker as reference

MCUdude, 2022-09-13:
I don't have my Windows computer available at the moment, so I can't send a SigmaStudio screenshot.

However, this _should_ be relatively trivial to figure out yourself. If you just want a static DSP program that acts as a filter for a 3-way speaker that doesn't need to be adjusted in real-time, you don't really need an Arduino and this library, you can use an Analog Devices USBi programmer and flash an external i2c EEPROM directly that the DSP can selfboot from.

And the DSP program is a small part of the overall system. You would still need a decent power supply, preamps, and an amplifier capable of driving a speaker.

closed 2022-12-17 by MCUdude