MCUdude/SigmaDSP #35: Change FS while running

gotrune, 2022-12-17:
Hello, first of all a huge thanks for these joby ;)
Then i have a custom board with an ADAU1701, an esp12f as master, an stm32 as usb audio device and an si5153c as cmaster lock reference. 
The stm32 can change its own fs clock (in windows) and telling the esp12 the new FS
So the question is: how can i change the fs clock of the adau1701 in the code:
I've tried something like that:

`
SigmaDSP dsp(Wire, DSP_I2C_ADDRESS, 48000.00f /*,12*/);
SigmaDSP dsp2(Wire, DSP_I2C_ADDRESS, 44100.00f /*,12*/);

setup(){
            dsp.begin();
            dsp2.begin();
            loadProgram(dsp);
            }

switch (frame_rate){
			case 0 :
				pcf20.write(Pin_expander_ADAU,0);
				si5351.set_freq(1128960000ULL, SI5351_CLK1);
				delay(50);
				pcf20.write(Pin_expander_ADAU,1);
				delay(50);
				loadProgram(dsp2);
				
			break;

			case 1 :
				pcf20.write(Pin_expander_ADAU,0);
				si5351.set_freq(1228800000ULL, SI5351_CLK1);
				delay(50);
				pcf20.write(Pin_expander_ADAU,1);
				delay(50);
				loadProgram(dsp);	
			break;
                   }
`

MCUdude, 2022-12-17:
I'm pretty sure you can't change the sample rate on the fly on the ADAU1701, it's compiled into the DSP firmware. However, you could have two different firmware that you can switch between. I've never tried it, but it may work with the library out of the box.

The sample rate passed though the constructor is only used when calculating certain filter values, and does not set or change the DSP sample rate at all.

You would need two separate SigmaDSP parameter files, for instance, `SigmaDSP_parameters.h` and `SigmaDSP_parameters2.h`. You can modify SigmaDSP_paramters2.h of  so that you can call `loadProgram2(dsp)` or `loadProgram2(dsp2)`. Just make sure to rename all arrays in the file.

gotrune, 2022-12-17:
Ohhh ok i don't tjink about it. Thanks for answer rapidity, i'll test that and i tell you ;)

gotrune, 2022-12-18:
Hello,
Test that and work really fine, thanks a lot.
Just have to deal with a define at the begining of the paramaters.h and just remove it in the second program
   #ifndef SIGMADSP_PARAMETERS_H
   #define SIGMADSP_PARAMETERS_H

You said:
"The sample rate passed though the constructor is only used when calculating certain filter values, and does not set or change the DSP sample rate at all" 
Yes i see that, my 200hz filter is now about 183hz...
The declaration of multiple constructor make my program more complicated..

My question is: is any easy way to change te fs's constructor ?
    SigmaDSP dsp(Wire, DSP_I2C_ADDRESS, 48000.00f /*,12*/);

MCUdude, 2022-12-18:
> My question is: is any easy way to change te fs's constructor ?

The library isn't really designed to be able to change the sample rate on the fly.
Remove `const` from this line. Now you can change the sample rate used for calculation on the fly by writing to `dsp.FS`.

gotrune, 2023-03-22:
Hi, digging arround 96khz, i find something weird in the sigmadsp.cpp at line 1011-1012
`   addr[0] = 0x00;
    addr[1] = 0x3C; // Set the IST bit (initiate safeload transfer bit)
 `
Wich set the core register at 48khz, i made a litle modification like:

`addr[0] = readRegister(dspRegister::CoreRegister,1);
 addr[1] = readRegister(dspRegister::CoreRegister,2) + 0b00100000; // Set the IST bit (initiate safeload transfer bit)
`
And then only modify the safeload bit :)
Bye

closed 2023-09-14 by MCUdude