MCUdude/SigmaDSP #37: Failing when compile with for ESP32

logicsb, 2023-10-28:
Hi, i've tried use the library with a ESP32 S3 and i'm got some compile erros of override ambiguity. I think the @define that change the parameter from uint16_t to int is unnecessary and without then the code compile successfully. 

This code causes ambiguity error on call of the function "safeload_writeRegister" at line: 184
`
#if defined(ARDUINO_ARCH_ESP8266) || defined(ARDUINO_ARCH_ESP32)
  void safeload_writeRegister(uint16_t memoryAddress, int16_t data, bool finished);
#else
  void safeload_writeRegister(uint16_t memoryAddress,     int data, bool finished);
#endif
`
Other question:
Using that library is capable of to read the signal in realtime to create a level meter with a Fourier transform?

Thanks so much.

MCUdude, 2023-11-20:
Hi!

I've used the library with the "regular" ESP32 without any issues. Can you post the entire error from the compiler, so I have a chance to figure out what's wrong?

> I think the define that changes the parameter from uint16_t to int is unnecessary and without then the code compiles successfully.

IIRC I had to do this to make it compile on STM32's.

> Using that library is capable of to read the signal in realtime to create a level meter with a Fourier transform?

You can read the signal in real-time, but I can't guarantee how it will perform. For every time you read, you'll first have to send a read request, and then the state can be read back. I've done it for signal detection reasons, but never for creating a Fourier transform.

MCUdude, 2023-11-20:
I'm not getting any compiler warnings or errors when compiling for the ESP32 S3. Are you sure you have installed the latest v1.1.2 version of the SigmaDSP library?

MCUdude, 2023-11-20:
Does it work if you use c++11, c++14 or c++17?

logicsb, 2023-11-21:
Same error using that versions, probably some flags difference i quess. But. I think the conversion from int to uint16_t can be removed without any problems. I don't know about the ESP8266 cuz i never worked with then but with any ESP32 chip will be fine.

MCUdude, 2023-11-21:
Can you upload your PlatformIO project as a ZIP file, so I can try to figure out _why_ you're getting this error in the first place? I do understand that the fix for your particular hardware is simple, but I can't risk breaking compatibility for other hardware.

logicsb, 2023-11-21:
Sure. Here is
[ESP32.zip](https://github.com/MCUdude/SigmaDSP/files/13427537/ESP32.zip)

MCUdude, 2023-11-21:
That's quite a lineup of platform packages?
But how do I install the gcc compiler? I'm getting the following error:

```
$ pio run -e nbox
Processing nbox (platform: espressif32; board: esp32-s3; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3.html
PLATFORM: Espressif 32 (1.12.3+sha.1c76b1b) > ESP32 S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.7438c6e 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.5c61c89308.sha.803092e 
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-esptoolpy @ 1.20600.0 (2.6.0) 
 - tool-openocd-esp32 @ 1.1000.20190709 (10.0) 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208 
 - toolchain-xtensa32 @ 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit APDS9960 Library @ 1.2.5
|-- INA226_WE @ 1.2.8
|-- TCA9548A @ 1.1.3
|-- ModbusMaster @ 2.0.1
|-- Adafruit MCP23017 Arduino Library @ 2.3.2
|-- SigmaDSP @ 1.1.2
|-- ZMPT101B @ 1.0.1+sha.c0559f6
|-- Wire @ 2.0.0
|-- SD @ 2.0.0
Building in debug mode
Compiling .pio/build/nbox/src/Audio.cpp.o
sh: xtensa-esp32s3-elf-g++: command not found
Compiling .pio/build/nbox/src/BatteryCharger.cpp.o
sh: xtensa-esp32s3-elf-g++: command not found
Compiling .pio/build/nbox/src/DigiPot/DigiPot.cpp.o
sh: xtensa-esp32s3-elf-g++: command not found
*** [.pio/build/nbox/src/Audio.cpp.o] Error 127
Compiling .pio/build/nbox/src/EventManager.cpp.o
*** [.pio/build/nbox/src/BatteryCharger.cpp.o] Error 127
*** [.pio/build/nbox/src/DigiPot/DigiPot.cpp.o] Error 127
sh: xtensa-esp32s3-elf-g++: command not found
*** [.pio/build/nbox/src/EventManager.cpp.o] Error 127
======================================================= [FAILED] Took 2.60 seconds =======================================================

```

logicsb, 2023-11-21:
Yeah, that's sucks. I got a bunch of problem to use de new IDF version with new ADC library + Arduino Framework. I'm trying to remember how i fixed this issue.

logicsb, 2023-11-21:
Open PlatformIO Core CLI and try to run: pio pkg update 
It's probably will install the compiler which is specified in platformio.ini

MCUdude, 2023-11-21:
Thanks. I can confirm the issue you're having.

It's an easy fix though.

In SigmaDSP.cpp, find/replace the section with this:

```cpp

#if defined(ARDUINO_ARCH_ESP8266) || defined(ARDUINO_ARCH_ESP32)
  void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress, int16_t data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
#endif
void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,       int data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,  uint32_t data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,  uint16_t data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,   uint8_t data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,    double data, bool finished) { safeload_writeRegister(memoryAddress,   (float)data, finished); }

```

and do the same thing in SigmaDSP.h, so the int16_t variant is _only_ available for the ESP8266/ESP32.
I'll test a little more testing and push a fix soon.

logicsb, 2023-11-22:
Thanks so mutch. 
About the other question. Using SigmaDSP is capable of to read the signal in realtime to create a level meter with a Fourier transform?

MCUdude, 2023-11-22:
> You can read the signal in real-time, but I can't guarantee how it will perform. For every time you read, you'll first have to send a read request, and then the state can be read back. I've done it for signal detection reasons, but never for creating a Fourier transform.

logicsb, 2023-11-22:
Thanks, i'll try read directly from the DAC board using analog read.

MCUdude, 2023-11-22:
You can always try to read the signal over i2c, just to see how it performs. I would bump the i2c speed up to 400kHz though.
Use the `dsp.readBack` function. and a readback block in Sigma Studio.

logicsb, 2023-11-22:
I'll test it. I have an QCC5171 chip that i had programmed the firmware and can i use the Kalimba's native Qualcomm DSP to send the signal over i2c to ESP or analog signal from the DAC (ES9038Q2M). I've 3 separated microcontrollers. Bluetooth (QCC5171), DAC (ES9038Q2M) and de DSP (ADAU1701), probably will perform better.

qniens, 2024-06-21:
I'm having the same issues when compiling the Template example. For a ESP32.

```
`c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.cpp: In member function 'void SigmaDSP::mux(uint16_t, uint8_t, uint8_t)':
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.cpp:97:29: error: call of overloaded 'safeload_writeRegister(uint16_t, int, bool)' is ambiguous
   97 |       safeload_writeRegister(startMemoryAddress++, 0x00, false);
      |       ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.cpp:1:
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:157:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint8_t*, bool)'
  157 |     void safeload_writeRegister(uint16_t memoryAddress,  uint8_t *data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:158:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, int32_t, bool)'
  158 |     void safeload_writeRegister(uint16_t memoryAddress,   int32_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:159:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, float, bool)'
  159 |     void safeload_writeRegister(uint16_t memoryAddress,     float data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:161:12: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, int16_t, bool)'
  161 |       void safeload_writeRegister(uint16_t memoryAddress, int16_t data, bool finished);
      |            ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:165:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint32_t, bool)'
  165 |     void safeload_writeRegister(uint16_t memoryAddress,  uint32_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:166:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint16_t, bool)'
  166 |     void safeload_writeRegister(uint16_t memoryAddress,  uint16_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:167:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint8_t, bool)'
  167 |     void safeload_writeRegister(uint16_t memoryAddress,   uint8_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:168:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, double, bool)'
  168 |     void safeload_writeRegister(uint16_t memoryAddress,    double data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.cpp:100:27: error: call of overloaded 'safeload_writeRegister(uint16_t, int, bool)' is ambiguous
  100 |     safeload_writeRegister(startMemoryAddress++, 0x00800000, (i == numberOfIndexes ? true : false));
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:158:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, int32_t, bool)'
  158 |     void safeload_writeRegister(uint16_t memoryAddress,   int32_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:159:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, float, bool)'
  159 |     void safeload_writeRegister(uint16_t memoryAddress,     float data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:161:12: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, int16_t, bool)'
  161 |       void safeload_writeRegister(uint16_t memoryAddress, int16_t data, bool finished);
      |            ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:165:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint32_t, bool)'
  165 |     void safeload_writeRegister(uint16_t memoryAddress,  uint32_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:166:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint16_t, bool)'
  166 |     void safeload_writeRegister(uint16_t memoryAddress,  uint16_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:167:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, uint8_t, bool)'
  167 |     void safeload_writeRegister(uint16_t memoryAddress,   uint8_t data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.h:168:10: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, double, bool)'
  168 |     void safeload_writeRegister(uint16_t memoryAddress,    double data, bool finished);
      |          ^~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
 1117 | void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,   uint8_t data, bool finished) { safeload_writeRegister(memoryAddress, (int32_t)data, finished); }
      |      ^~~~~~~~
c:\Users\quint\Documents\Arduino\libraries\SigmaDSP\src\SigmaDSP.cpp:1118:6: note: candidate: 'void SigmaDSP::safeload_writeRegister(uint16_t, double, bool)'
 1118 | void SigmaDSP::safeload_writeRegister(uint16_t memoryAddress,    double data, bool finished) { safeload_writeRegister(memoryAddress,   (float)data, finished); }
      |      ^~~~~~~~

exit status 1

Compilation error: exit status 1`
```

MCUdude, 2024-06-21:
@qniens Using Arduino IDE or PlatformIO?

I'm not getting any errors when compiling the 0_Template example for the ESP32 using Arduino IDE 1.8.19 and SigmaDSP v1.1.4.

[image]

qniens, 2024-06-21:
For the test I'm using the Arduino IDE 2.3.2 and the latest version of SigmaDSP
[image]

MCUdude, 2024-06-21:
Sorry, but I can't reproduce your error. I'm using Arduino 2.3.2 and Windows 11. Which ESP32 boards manager version are you using?

[image]

qniens, 2024-06-21:
I believe the issue stems from the recent update to Arduino version 3.0.1, which includes breaking changes in the code.
On PlatformIO I can compile the code without an issue

MCUdude, 2024-06-21:
@qniens using the ESP32 Arduino v3.0.1 package I could reproduce the issue. I'll push a fix for it soon.

Menawhile, you can replace

```c
#if defined(ARDUINO_ARCH_ESP8266) || defined(ARDUINO_ARCH_ESP32)
```

with 

```c
#if defined(ARDUINO_ARCH_ESP8266) || (defined(ARDUINO_ARCH_ESP32) && ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0))
```

in SigmaDSP.h and SigmaDSP.cpp, and it should compile again.

MCUdude, 2024-06-21:
A new SignaDSP release, v1.1.5 is now available though the Arduino Library manager and the PlatformIO registry. Thanks for reporting the bug!

closed 2024-06-21 by MCUdude