MCUdude/SigmaDSP #39: Cross-Fade Mixer Support / Single Control Mixer Support

DatanoiseTV, 2024-03-15:
It seems like the mixer elements are not supported (yet)?
I wanted to try crossfading between 2 channels via I2C.

MCUdude, 2024-03-15:
There are lots and lots of different blick I haven't reverse-engineered. But the neat thing about this library is that it's easy to create your own functions to control different blocks.

You can play around with the capture window in SigmaStudio to determine how to control a particular block.

Here, for instance, is the function to control a volume slider with slew. In your case, I would assume you could add two volume sliders, and merge them at the end using a mixer block.

https://github.com/MCUdude/SigmaDSP/blob/b0d8c8a0be2de6cfdd49f54354290962b7ba9608/src/SigmaDSP.cpp#L173-L188

WillyBilly06, 2024-07-05:
How can I fix the "bricked DSP" problem whenever I use the function "loadProgram(dsp)"? I am able to write to the EEPROM but not load the DSP in order for the DSP to work.