MCUdude/SigmaDSP #42: Unused parameter

qniens, 2024-07-23:
Hi

Thank you for the amazing work you've done on this project. It's been incredibly helpful.

I've been using the SigmaDSP class and noticed something that might be an oversight or perhaps I'm missing its purpose. In all the provided examples, the class is instantiated as follows:

`SigmaDSP dsp(Wire, DSP_I2C_ADDRESS, 48000.00f /*,12*/);`
However, upon inspecting the class implementation, it appears that the third parameter (48000.00f in this case) is never used within the class. 

Is this on purpose?

MCUdude, 2024-07-23:
The sample rate (`48000.00f`) is copied into the `FS` constant in the constructor. `FS`is used in various functions, such as `SigmaDSP::dynamicBass` and `SigmaDSP::audioDelay`.

qniens, 2024-07-23:
You're correct. Thank you for the quick response

closed 2024-07-23 by qniens