MCUdude/SigmaDSP #43: Limiter / Crossover Module?

PatrickMassler, 2024-08-06:
Hello, thats really a great Library you have written here, thank you!

Now on to my questions: is it possible to change settings of a RMS-Limiter-Block with your Library? also can Crossover-Values be adjusted? 

A quick example or Explanation would help me a lot to go on with my Project! 

Thanks in Advance,
Patrick

MCUdude, 2024-08-06:
Hi!

Anything can be changed using this library. It gives you full R/W access to the DSP. However, the tricky part is to "crack" the algorithm so that you're sending the correct values. Can you send a screenshot + a description of where I can find this exact block in Sigma Studio?

PatrickMassler, 2024-08-06:
Wow, that was a quick reply! :)

i attached two screenshots showing the Blocks i am most interested in,

Limiter-Block is in: Dynamics Processors -> RMS -> Limiter
Crossover-Block is in: Filters -> Crossover -> Single Precision / Double
Precision -> 3-Way -> Crossover

Thanks a Lot in advance!
Greetings from Germany,
Patrick

MCUdude, 2024-08-06:
I have not used these blocks myself, but everything is possible as long as you know what to send, and to which DSP addresses.
This library offers full, low level access to everything, but does not add support for every DSP block SIgma Studio offers.

I've used the the [``compressorPeak``](https://github.com/MCUdude/SigmaDSP/blob/ad35ebe7b3f35d3b56e82cdb055ca535708b6cbd/src/SigmaDSP.cpp#L932) function in this library with  
`Dynamics Processors -> Peak -> Standard Resolution -> Lower Range (-90 to +6dB) -> Post Gain -> Stereo -> No Ext Detector Input > Peak (gain)`.

I'm sure you can find a similar RMS compressor block that will work with [`compressorRMS`](https://github.com/MCUdude/SigmaDSP/blob/ad35ebe7b3f35d3b56e82cdb055ca535708b6cbd/src/SigmaDSP.cpp#L846C16-L846C29). And you can always just copy the function from the library, paste it in your sketch and modify it to fit your needs if necessary.

I have never used the three-way crossover, but it seems to me like you can use the same algorithm as used in [`EQsecondOrder`](https://github.com/MCUdude/SigmaDSP/blob/ad35ebe7b3f35d3b56e82cdb055ca535708b6cbd/src/SigmaDSP.cpp#L576) for each of the four controllers, if you're using Butterworth 12 or Bessel 12 in the crossover. And 12 means 12 dB or 2nd order.

So for the four "bands" use:
| Band 1 (Low) | Band 2 (Mid Low)| Band 3 (Mid High)| Band 4 (High)|
|-------|--|--|--|
| Butterworth 12 | Butterworth 12 | Butterworth 12 | Butterworth 12 |
| Bessel 12 | Bessel 12 | Bessel 12 | Bessel 12 | 

https://github.com/MCUdude/SigmaDSP/blob/ad35ebe7b3f35d3b56e82cdb055ca535708b6cbd/src/parameters.h#L13-L28

Play around with the Capture window in Sigma Studio to figure out what happens with the various addresses when changing the settings.

PatrickMassler, 2024-08-06:
So, If i understand you correctly, i could Just do a Safeload-Write to the
corresponding Address to Set the Value i desire? The conversion Formula for
the value would be pow(10, dB / 20)?
Greetings

MCUdude, 2024-08-06:
Yes, just use safeload_write. But you'll have to do some math first. Either by rolling your own function, or using the functions I mentioned in my previous post.

FYI I'll recommend you to reply on the Github page, and not by email. There may be things like embedded code and tables that may or may not be present in the email Github sends you.

PatrickMassler, 2024-08-07:
Hello, 

i quickly whipped up some Code yesterday for the Limiter Block:

`void SigmaDSP::dynamics_limiter(uint16_t startMemoryAddress, int threshold)
{
  //Limiter_sub
  //Threshold:  0x0095
  //RMS_TC:     0x0096
  //Decay:      0x0097 / 0x0098
  float threshold_values[] = {
  1.000000000000000, 0.891250967979431, 0.794328212738037, 0.707945823669434, 0.630957365036011, 0.562341332435608, 0.501187205314636, 0.446683645248413,
  0.398107171058655, 0.35481333732605,  0.316227793693542, 0.281838297843933, 0.251188635826111, 0.223872065544128, 0.199526190757751, 0.177827954292297,
  0.158489346504211, 0.141253709793091, 0.125892519950867, 0.112201809883118, 0.100000023841858, 0.0891250371932983,0.0794328451156616,0.0707945823669434,
  0.0630956888198853};
  int i = threshold * -1;
  safeload_write(startMemoryAddress, threshold_values[i]);
}`

Could you have a quick look if anything looks off? The Values of the array are directly taken from SigmaStudio

MCUdude, 2024-08-12:
Give it a try and see how it works!

> int i = threshold * -1;

THe compiler will probably optimize this, but why not do ``int i = -threshold`` instead?

I would probably use `safeload_write()`, instead, but it doesn't really matter.

closed 2025-07-03 by MCUdude